purdue-orbital/avionics #102: Design 12V to 5V converter

mchartigan, 2020-02-19:
The Raspberry Pi 3B+ requires a constant 5V power supply to operate as intended without undervolting.

Create a circuit diagram which takes a standard 12V input from a battery and converts it into a stable 5V output to microUSB (the microUSB output need not be included -- a positive and negative terminal will suffice for now).

This will eventually be integrated into a PCB assembly, likely using Kicad, so formatting to future-proof against this transition would be appreciated.

mchartigan, 2020-02-19:
Potential ways to make this easier -- I believe this is a common design, so you can likely find pre-existing solutions on the internet. Just translate those into kicad

ZecongLiang, 2020-02-20:
I agree with Mark. This 12V to 5V converter is commonly known as a buck converter and there are many designs out there.

closed 2021-01-12 by Cjberke