purdue-orbital/avionics #105: Generate a DS3231 Circuit Diagram in KiCad

mchartigan, 2020-02-19:
The avionics PCB will use embedded systems rather than a breakout board to interface with sensors.

A circuit diagram needs to be created in KiCad for the DS3231 Real-Time Clock. Use all available resources to complete this -- this clock is fairly common, so I would expect it to have pre-made circuit diagrams already.

Please keep an accurate parts list of components required for this subassembly.

mchartigan, 2020-02-19:
[Helpful link](https://datasheets.maximintegrated.com/en/ds/DS3231.pdf) to the product datasheet

ZecongLiang, 2020-02-20:
Here is a link to the DS3231 on Adafruit's website, which includes the Eagle files: https://learn.adafruit.com/adafruit-ds3231-precision-rtc-breakout/downloads

closed 2021-01-12 by Cjberke